heinekmadsen/esphome_components #6: Wrong icon for the battery sensors

mathias-jakobsen, 2022-01-24:
Usually a battery sensor has a dynamic icon based on the current percentage.
This is currently not the case with the wavin battery sensors as they have a fixed percentage icon.
Is it possible to change this?

ssieb, 2022-01-24:
You would have do something special on the HA side.  esphome can only send an icon suggestion to it and that can't be changed at runtime.

mathias-jakobsen, 2022-02-10:
Wouldn't HA be able to figure out the correct icons, if you only define the device class and omit the icon suggestion?

Line 29 in:
https://github.com/heinekmadsen/esphome_components/tree/main/components/wavinAhc9000/climate/__init__.py
`
cv.Optional(CONF_BATTERY_LEVEL): sensor.sensor_schema(UNIT_PERCENT, ICON_PERCENT, 0, DEVICE_CLASS_BATTERY)
`

to

`
cv.Optional(CONF_BATTERY_LEVEL): sensor.sensor_schema(unit_of_measurement=UNIT_PERCENT,  accuracy_decimals=0, device_class=DEVICE_CLASS_BATTERY)
`

heinekmadsen, 2022-03-21:
Works fine with Wavin Ahc 9000 v2 component

closed 2022-03-21 by heinekmadsen