heinekmadsen/esphome_components #9: V2 - Channel 9 and 10 is showing temperature at 0 degrees celcius

mathias-jakobsen, 2022-03-21:
I just set up the V2 of the wavin component. It all works splendid, except for channel 9 and 10, which always shows the current temperature at 0 degrees celcius. See picture below.
The climate control works as expected.

ESPHome configuration:
```
substitutions:
  device: "wavin"
  name: "Wavin"
  log_level: DEBUG

  channel_01_friendly_name: "Kitchen-Dining Room"
  channel_02_friendly_name: "Guest Toilet"
  channel_03_friendly_name: "test1"
  channel_04_friendly_name: "test2"
  channel_05_friendly_name: "Bedroom"
  channel_06_friendly_name: "Walk In"
  channel_07_friendly_name: "Walkthrough"
  channel_08_friendly_name: "Bathroom"
  channel_09_friendly_name: "Playroom"
  channel_10_friendly_name: "Child's Room"
  channel_11_friendly_name: "test4"
  channel_12_friendly_name: "test5"
  channel_13_friendly_name: "test6"
  channel_14_friendly_name: "test7"
  channel_15_friendly_name: "test8"
  channel_16_friendly_name: "test9"
  
  channel_01_id: kitchen_dining_room
  channel_02_id: guest_toilet
  channel_03_id: test1
  channel_04_id: test2
  channel_05_id: bedroom
  channel_06_id: walk_in
  channel_07_id: walkthrough
  channel_08_id: bathroom
  channel_09_id: playroom
  channel_10_id: childs_room
  channel_11_id: test4
  channel_12_id: test5
  channel_13_id: test6
  channel_14_id: test7
  channel_15_id: test8
  channel_16_id: test9

  ssid_set: <redacted>
  password_set: <redacted>

esphome:
  name: ${device}
  platform: ESP8266
  board: esp01_1m
  project:
    name: "home.wavin_gulvvarme"
    version: "1.0.0"

logger:
  baud_rate: 0
    
api:
  
ota:
  password: <redacted>

wifi:
  ssid: ${ssid_set}
  password: ${password_set}

  manual_ip:
    static_ip: <redacted>
    gateway: <redacted>
    subnet: <redacted>

  ap:
    ssid: <redacted>
    password: <redacted>

captive_portal:  

packages:
  remote_package:
    url: https://github.com/heinekmadsen/esphome_components
    ref: main
    files: 
      - components/wavinahc9000v2/configs/channel_01.yaml 
      - components/wavinahc9000v2/configs/channel_02.yaml 
      #- components/wavinahc9000v2/configs/channel_03.yaml 
      #- components/wavinahc9000v2/configs/channel_04.yaml 
      - components/wavinahc9000v2/configs/channel_05.yaml 
      - components/wavinahc9000v2/configs/channel_06.yaml 
      - components/wavinahc9000v2/configs/channel_07.yaml 
      - components/wavinahc9000v2/configs/channel_08.yaml 
      - components/wavinahc9000v2/configs/channel_09.yaml 
      - components/wavinahc9000v2/configs/channel_10.yaml 
      #- components/wavinahc9000v2/configs/channel_11.yaml 
      #- components/wavinahc9000v2/configs/channel_12.yaml
      #- components/wavinahc9000v2/configs/channel_13.yaml 
      #- components/wavinahc9000v2/configs/channel_14.yaml
      #- components/wavinahc9000v2/configs/channel_15.yaml 
      #- components/wavinahc9000v2/configs/channel_16.yaml 
    refresh: 0s
  
modbus:
  - id: ${device}_modbus
    uart_id: uart_${device}

uart:
  - id: uart_${device}
    rx_pin: RX
    tx_pin: TX
    baud_rate: 38400
    stop_bits: 1
    parity: NONE
    
modbus_controller:
  id: ${device}_modbus_controller
  address: 1
  modbus_id: wavin_modbus
  update_interval: 5s
```

[image]

mathias-jakobsen, 2022-03-21:
No, it is two separate rooms. They each have their own room thermostat.

In another section of the house, I have 3 strings sharing one room thermostat with no issues.

heinekmadsen, 2022-03-21:
Not really sure why then :(

mathias-jakobsen, 2022-03-22:
Ah that is too bad :(

briis, 2022-08-18:
I see the same. I have 13 Thermostats, the first 10 I add work fine (circuit 1 to 10), but 11, 12 and 13 have 0 in current temp and they are separate thermostats in separate rooms.

heinekmadsen, 2022-10-19:
I'm aware of the issue, but not seeing it on my own system, so not really able to troubleshoot :(

exetico, 2024-05-04:
I've moved from the MQTT solution to this (v2, ESPHome), and I'm seeing the same.

I've also noticed that a few of the thermostats states are reported for the wrong one. Most works OK, but not all of them. Can I do something for this to be fixed? 

Here's a log from one interval data-fetch execution:

<details>
  <summary>See logs from modbus data fetch (9 channels)</summary>
 
```
[22:49:58][D][modbus_controller.sensor:025]: Sensor new state: 2.00
[22:49:58][D][sensor:094]: 'wavin Battery Office (Room 2)': Sending state 20.00000 % with 1 decimals of accuracy
[22:49:58][D][main:123]: MODE for Channel 01 is: 0
[22:49:59][D][main:123]: MODE for Channel 01 is: 0
[22:49:59][D][modbus.number:023]: Number new state : 21.00
[22:49:59][D][number:012]: 'wavin Target Temperature Entrance': Sending state 21.000000
[22:49:59][D][climate:396]: 'wavin Entrance' - Sending state:
[22:49:59][D][climate:399]:   Mode: AUTO
[22:49:59][D][climate:401]:   Action: IDLE
[22:49:59][D][climate:419]:   Current Temperature: 21.90°C
[22:49:59][D][climate:425]:   Target Temperature: 21.00°C
[22:49:59][D][modbus.number:023]: Number new state : 20.00
[22:49:59][D][number:012]: 'wavin Target Temperature Guestroom (Room 1)': Sending state 20.000000
[22:49:59][D][climate:396]: 'wavin Guestroom (Room 1)' - Sending state:
[22:49:59][D][climate:399]:   Mode: AUTO
[22:49:59][D][climate:401]:   Action: IDLE
[22:49:59][D][climate:419]:   Current Temperature: 21.80°C
[22:49:59][D][climate:425]:   Target Temperature: 20.00°C
[22:49:59][D][modbus.number:023]: Number new state : 22.00
[22:49:59][D][number:012]: 'wavin Target Temperature Office (Room 2)': Sending state 22.000000
[22:49:59][D][climate:396]: 'wavin Office (Room 2)' - Sending state:
[22:49:59][D][climate:399]:   Mode: AUTO
[22:49:59][D][climate:401]:   Action: IDLE
[22:49:59][D][climate:419]:   Current Temperature: 21.50°C
[22:49:59][D][climate:425]:   Target Temperature: 22.00°C
[22:49:59][D][main:123]: MODE for Channel 01 is: 0
[22:49:59][D][modbus_controller.sensor:025]: Sensor new state: 2.00
[22:49:59][D][sensor:094]: 'wavin Battery Livingroom': Sending state 20.00000 % with 1 decimals of accuracy
[22:49:59][D][modbus_controller.sensor:025]: Sensor new state: 219.00
[22:49:59][D][sensor:094]: 'wavin Temperature Entrance': Sending state 21.90000 °C with 1 decimals of accuracy
[22:49:59][D][climate:396]: 'wavin Entrance' - Sending state:
[22:49:59][D][climate:399]:   Mode: AUTO
[22:49:59][D][climate:401]:   Action: IDLE
[22:49:59][D][climate:419]:   Current Temperature: 21.90°C
[22:49:59][D][climate:425]:   Target Temperature: 21.00°C
[22:49:59][D][modbus.number:023]: Number new state : 21.50
[22:49:59][D][number:012]: 'wavin Target Temperature Livingroom': Sending state 21.500000
[22:49:59][D][climate:396]: 'wavin Livingroom' - Sending state:
[22:49:59][D][climate:399]:   Mode: AUTO
[22:49:59][D][climate:401]:   Action: IDLE
[22:49:59][D][climate:419]:   Current Temperature: 22.80°C
[22:49:59][D][climate:425]:   Target Temperature: 21.50°C
[22:49:59][D][main:123]: MODE for Channel 01 is: 0
[22:49:59][D][modbus_controller.sensor:025]: Sensor new state: 2.00
[22:49:59][D][sensor:094]: 'wavin Battery Entrance': Sending state 20.00000 % with 1 decimals of accuracy
[22:49:59][D][main:123]: MODE for Channel 01 is: 0
[22:49:59][D][modbus.number:023]: Number new state : 20.00
[22:49:59][D][number:012]: 'wavin Target Temperature Livingroom Zone 2': Sending state 20.000000
[22:49:59][D][climate:396]: 'wavin Livingroom Zone 2' - Sending state:
[22:49:59][D][climate:399]:   Mode: AUTO
[22:49:59][D][climate:401]:   Action: IDLE
[22:49:59][D][climate:419]:   Current Temperature: 22.80°C
[22:49:59][D][climate:425]:   Target Temperature: 20.00°C
[22:49:59][D][main:123]: MODE for Channel 01 is: 0
[22:49:59][D][modbus_controller.sensor:025]: Sensor new state: 245.00
[22:49:59][D][sensor:094]: 'wavin Temperature Kitchen': Sending state 24.50000 °C with 1 decimals of accuracy
[22:49:59][D][climate:396]: 'wavin Kitchen' - Sending state:
[22:49:59][D][climate:399]:   Mode: AUTO
[22:49:59][D][climate:401]:   Action: IDLE
[22:49:59][D][climate:419]:   Current Temperature: 24.50°C
[... 72 lines cut ...]
[22:50:00][D][main:123]: MODE for Channel 01 is: 0
[22:50:00][D][main:123]: MODE for Channel 09 is: 0
[22:50:00][D][modbus.number:023]: Number new state : 18.50
[22:50:00][D][number:012]: 'wavin Target Temperature Utilityroom': Sending state 18.500000
[22:50:00][D][climate:396]: 'wavin Utilityroom' - Sending state:
[22:50:00][D][climate:399]:   Mode: AUTO
[22:50:00][D][climate:401]:   Action: IDLE
[22:50:00][D][climate:419]:   Current Temperature: 0.00°C
[22:50:00][D][climate:425]:   Target Temperature: 18.50°C
[22:50:01][D][modbus_controller.sensor:025]: Sensor new state: 7.00
[22:50:01][D][sensor:094]: 'wavin Battery Bedroom': Sending state 70.00000 % with 1 decimals of accuracy
[22:50:01][D][main:123]: MODE for Channel 01 is: 0
[22:50:01][D][modbus.number:023]: Number new state : 21.50
[22:50:01][D][number:012]: 'wavin Target Temperature Restroom': Sending state 21.500000
[22:50:01][D][climate:396]: 'wavin Restroom' - Sending state:
[22:50:01][D][climate:399]:   Mode: AUTO
[22:50:01][D][climate:401]:   Action: IDLE
[22:50:01][D][climate:419]:   Current Temperature: 24.80°C
[22:50:01][D][climate:425]:   Target Temperature: 21.50°C
[22:50:01][D][modbus_controller.sensor:025]: Sensor new state: 4.00
[22:50:01][D][sensor:094]: 'wavin Battery Kitchen': Sending state 40.00000 % with 1 decimals of accuracy

```
  
</details>