heinekmadsen/esphome_components #15: Kitchen uses 2 channels, but only has a wall sensor on 1

dkgof, 2022-05-29:
Our kitchen use both channel 5 and 6 for the floor heating, but only channe 5 has a wall mounted controller/sensor.

This means that when using the Wavin components, all channels after 5, has the temperature offset by 1.

Example:

Channel 7's temperature is showing up as temperature for channel 6 in HomeAssistant.

Im not sure why it is made like this, or what the best fix is.

mathias-jakobsen, 2022-10-09:
I have the same issue. Any update on this?

heinekmadsen, 2022-10-20:
@dkgof Do you have alle the config files in your yaml?
I have two strings sharing the same room thermostat. The string are connected to channel  7 and 8.
My climate entity for channel 8 just reads 0 in current temp and 20 in target temp, which is expected as i doesn't have a sensor paired.

Can you share your yaml?

lynx000, 2022-10-20:
Thank you for looking into this issue Heine. I know it's a difficult one. I am also willing to help troubleshoot. Let my know if you want to setup a session

heinekmadsen, 2022-10-20:
@lynx000 Start by sharing your yaml and describe which strings are sharing sensors

dkgof, 2022-10-20:
I ended up forking the repo, and changing the channel that the temperature sensor is fetched from, on all channels past the "dummy" channel in the kitchen.

Here is my yaml:
https://pastebin.com/bsUvPYhJ

And here is the forked repo:
https://github.com/dkgof/esphome_components

With these changes, the temperature is now correctly lined up with the correct channels.

Not sure if there is a better or more correct way of handling it.

lynx000, 2022-10-20:
yaml:
https://pastebin.com/YRJx1LB3

**Room and temp**

Here are my discoveries

**Garage**

Kredsløb 1
Termostat sidder på channel 1
Sensor passer til channel 1

**Bryggers**

Kredsløb 2,3,4
Termostat sidder på channel 3
Sensor passer til ???

**Almas_sovevaerelse**

Kredsløb 5
Termostat sidder på channel 5
Sensor passer til  channel 3

**Almas_legevaerelse**

Kredsløb 6
Termostat sidder på channel 6
Sensor passer til  channel 5

**Koekken-alrum**

Kredsløb 7,10,11,12
Termostat sidder på channel 7
Sensor passer til channel ????

**Badevaerelse 1**

Kredsløb 8
Termostat sidder på channel 8
Sensor passer til  channel 6

**Sovevaerelse**

Kredsløb 9
Termostat sidder på channel 9
Sensor passer til channel 7

**Badevaerelse 2**

Kredsløb 13
Termostat sidder på channel 13
Sensor passer til  channel 8

**Kontor**

Kredsløb 14
Termostat sidder på channel 14
Sensor passer til ???

briis, 2022-10-20:
This fixes my problem. Now I just need to figure out the correct mapping of ID's 😁 But all have now a Current Temperature

dkgof, 2022-10-21:
Unfortunately this does not work for me.

The way i fixed my problem, in my fork:

Channels 1, 2, 3, 4, 5 work as normal, because they come before the "extra" channel 6, that has no sensor associated.

Channels 7,8,9,10,11 need to have their battery and temperature address offset by 1, but still send to the old address for setting temperature.

I have tested that the correct physical room controller gets the new temperature set correcly like this, and that all rooms sensed temperatures match up.

Using the new channel substitution, i tried:

```# CHANNEL Alignments (ONLY change if you have offsets in channel vs temperature. Some have reported this if using multiple strings pr. room thermostat)
  channel_01: "0x00"
  channel_02: "0x01"
  channel_03: "0x02"
  channel_04: "0x03"
  channel_05: "0x04"
  channel_06: "0x04"
  channel_07: "0x06"
  channel_08: "0x07"
  channel_09: "0x08"
  channel_10: "0x09"
  channel_11: "0x0A"
```

Since my channel 5 and 6 are the shared ones, but all the ones after 6, have sensed temperature and battery offset, and temperature control still works.

lynx000, 2022-10-21:
Hello again. I managed to get the temperature working following your guide. Unfortunately then target temperature is now mixed up. The climate adjusts the wrong thermostats now :) 

Here are my new yaml:
https://pastebin.com/Ry0BS0Sj

dkgof, 2022-10-22:
I think that is the same problem i am facing.

The issue seems to be that the channels that has no physical room sensor, still takes up a channel, but not on temp sensor and battery sensor, so each time a channel is used without a room sensor, the following temp/battery sensor channel is offset by -1.

I recently helped my neighbour setup his wavin9000 in esphome, and he had 2 rooms with double channel, and after the first room, we needed to offset temp/battery by -1 channel, and after the second double channel room, we needed to offset the rest of the temp/battery sensors channels by -2.

I also removed the temp/battery sensor from the channels that did not have them, since they have no use.

dkgof, 2022-10-22:
I created a pull request with one possible way of solving the issue, not sure if its a good way.

https://github.com/heinekmadsen/esphome_components/pull/21

lynx000, 2022-10-22:
> This fixes my problem. Now I just need to figure out the correct mapping of ID's 😁 But all have now a Current Temperature

Are your target temperatures in your climate units current efter this fix ?

briis, 2022-10-23:
No, I was too fast. It fixed the issue with 0 current temperatures, but the rest then got messed up. So not a solution.

dkgof, 2022-10-23:
That is what my PR is trying to take care of, by just moving the sensed temperatures, without making the rest be off.

briis, 2022-10-23:
Thanks @dkgof I am looking forward to it being merged.

heinekmadsen, 2022-10-23:
Not really sure why you want those dummy sensors? What the purpose of having them in HA?
Aren't the climate entities what's you want? And a climate entity for each room sensor must be the goal?

dkgof, 2022-10-24:
They might not be needed, but it seems as the wavin9000 does reply with some data for those channels, so i figured you might want to see fx. the output sensor, to see if it turns on. But i can also see the reason to just skip them, and only use the temperature offsets.

dkgof, 2022-10-24:
I updated the pull request to not include the dummy channel configs.

briis, 2022-11-06:
Using @dkgof's yaml files my system is now working as expected. I have 13 circuits but only 10 thermostats, unfortunately the rule about offset with a specific number after the channels that were bundled together did not work for me. I had to swap around the sensor offsets in a non-logical way, but everything is now working. Temperature for the rooms are correct and I can control each thermostat. Here is my channel setup. If anyone can find a logic in this, I am happy to hear that.

```yaml
# CHANNEL friendly names (If using spaces remember to add ")
  channel_01_friendly_name:  "Bryggers"
  channel_02_friendly_name:  "Bjarnes Kontor"
  channel_03_friendly_name:  "Stue/Køkken"
  channel_04_friendly_name:  "Stue/Køkken"
  channel_05_friendly_name:  "Stue/Køkken"
  channel_06_friendly_name:  "Stue/Køkken"
  channel_07_friendly_name:  "Soveværelse"
  channel_08_friendly_name:  "Badeværelse"
  channel_09_friendly_name:  "Walkin"
  channel_10_friendly_name: "Daisys Kontor"
  channel_11_friendly_name: "Gæsteværelse"
  channel_12_friendly_name: "Entre"
  channel_13_friendly_name: "Gæstebadeværelse"
  
  # CHANNEL ID´S (ONLY LOWER CASE LETTERS, NO SPACES) used for entity ids
  channel_01_id: utility_room
  channel_02_id: office_bjarne
  #channel_03_id: living_room_kitchen
  #channel_04_id: living_room_kitchen
  #channel_05_id: living_room_kitchen
  channel_06_id: living_room_kitchen
  channel_07_id: bedroom
  channel_08_id: bathroom
  channel_09_id: walkin_closet
  channel_10_id: office_daisy
  channel_11_id: guest_room
  channel_12_id: hallway
  channel_13_id: guest_bathroom

 # CHANNEL Alignments (ONLY change if you have offsets in channel vs temperature. Some have reported this if using multiple strings pr. room thermostat)
  channel_01_sensor: "0x00"
  channel_02_sensor: "0x01"
#  channel_03_sensor: "0x02"
#  channel_04_sensor: "0x03"
#  channel_05_sensor: "0x04"
  channel_06_sensor: "0x06"
  channel_07_sensor: "0x07"
  channel_08_sensor: "0x02"
  channel_09_sensor: "0x03"
  channel_10_sensor: "0x09"
  channel_11_sensor: "0x04"
  channel_12_sensor: "0x08"
  channel_13_sensor: "0x05"

  channel_01: "0x00"
  channel_02: "0x01"
#  channel_03: "0x02"
#  channel_04: "0x03"
#  channel_05: "0x04"
  channel_06: "0x05"
  channel_07: "0x06"
  channel_08: "0x07"
  channel_09: "0x08"
  channel_10: "0x09"
  channel_11: "0x0A"
  channel_12: "0x0B"
  channel_13: "0x0C"
````

lynx000, 2022-11-12:
> Using @dkgof's yaml files my system is now working as expected. I have 13 circuits but only 10 thermostats, unfortunately the rule about offset with a specific number after the channels that were [...]

@briis did you pull config from Heine and just added the alignments ? I'm able to align the thermostats, but the channel sensor alignment does not seem to do anything for me. It's compiling just fine, it just does not change anything. :)

briis, 2022-11-12:
No, you need to use the files from @dkgof as they contain the changes for sensors. There is a PR open for that change, but that has not been merged. Here is how the package section looks like, not the url and dev lines:

```yaml
packages:
  remote_package:
    url: https://github.com/dkgof/esphome_components
    ref: dev
    files: 
      - components/wavinahc9000v2/configs/basic.yaml # REQUIRED FOR WAVIN AHC 9000
      - components/wavinahc9000v2/configs/channel_01.yaml
      - components/wavinahc9000v2/configs/channel_02.yaml
      #- components/wavinahc9000v2/configs/channel_03.yaml
      #- components/wavinahc9000v2/configs/channel_04.yaml
      #- components/wavinahc9000v2/configs/channel_05.yaml
      - components/wavinahc9000v2/configs/channel_06.yaml
      - components/wavinahc9000v2/configs/channel_07.yaml
      - components/wavinahc9000v2/configs/channel_08.yaml
      - components/wavinahc9000v2/configs/channel_09.yaml
      - components/wavinahc9000v2/configs/channel_10.yaml
      - components/wavinahc9000v2/configs/channel_11.yaml
      - components/wavinahc9000v2/configs/channel_12.yaml
      - components/wavinahc9000v2/configs/channel_13.yaml
    refresh: 0s
````

lynx000, 2022-11-12:
> No, you need to use the files from @dkgof as they contain the changes for sensors. There is a PR open for that change, but that has not been merged. Here is how the package section looks like, not [...]

Very cool. Thank you. Mine also works now.. ! :D

heinekmadsen, 2022-11-12:
Well... i have the alignment in the dev branch... just havnet had time to get it into main...

Hopefully I'll have some spare time in the coming week

Berjastein, 2023-01-12:
Im having simluar issues, how did you guys solve it?

This is my current setup, all the units with name "Alrúm" are 1 sensor.

`substitutions:
  # UNIQUE NAME FOR THE DEVICE
  device: wavin # Case sensitive!!!
  name: Wavin   # "Friendly name" - not case sensitive!!!
  
  # CHANNEL friendly names (If using spaces remember to add ")
  channel_01_friendly_name: "Alrúm 0"
  channel_02_friendly_name: "Stóra Vesi"
  channel_03_friendly_name: "Alrúm 1"
  channel_04_friendly_name: "Bryggers"
  channel_05_friendly_name: "Kamar Lóa"
  channel_06_friendly_name: "Alrúm 2"
  channel_07_friendly_name: "Okkara kamar"
  channel_08_friendly_name: "Alrúm 3"
  channel_09_friendly_name: "Alrúm 4"
  channel_10_friendly_name: "Alrúm 5"
  channel_11_friendly_name: "Kamar Vár"
  channel_12_friendly_name: "Lítla vesi"
  channel_13_friendly_name: "Channel 13"
  channel_14_friendly_name: "Channel 14"
  channel_15_friendly_name: "Channel 15"
  channel_16_friendly_name: "Channel 16"
  
  # CHANNEL ID´S (ONLY LOWER CASE LETTERS, NO SPACES) used for entity ids
  channel_01_id: alrum_0
  channel_02_id: stora_vesi
  channel_03_id: alrum_1
  channel_04_id: bryggers
  channel_05_id: kamar_loa
  channel_06_id: alrum_2
  channel_07_id: okkara_kamar
  channel_08_id: alrum_3
  channel_09_id: alrum_4
  channel_10_id: alrum_5
  channel_11_id: kamar_var
  channel_12_id: litla_vesi
  channel_13_id: channel_13
  channel_14_id: channel_14
  channel_15_id: channel_15
  channel_16_id: channel_16

# Uncomment the active channels on your Wavin Ahc 9000
packages:
  remote_package:
    url: https://github.com/heinekmadsen/esphome_components
    ref: main
    files: 
      - components/wavinahc9000v2/configs/basic.yaml # REQUIRED FOR WAVIN AHC 9000
      - components/wavinahc9000v2/configs/channel_02.yaml
      - components/wavinahc9000v2/configs/channel_03.yaml
      - components/wavinahc9000v2/configs/channel_04.yaml 
      - components/wavinahc9000v2/configs/channel_05.yaml 
      - components/wavinahc9000v2/configs/channel_06.yaml 
      - components/wavinahc9000v2/configs/channel_07.yaml 
      - components/wavinahc9000v2/configs/channel_08.yaml 
      - components/wavinahc9000v2/configs/channel_09.yaml 
      - components/wavinahc9000v2/configs/channel_10.yaml 
      - components/wavinahc9000v2/configs/channel_11.yaml 
      - components/wavinahc9000v2/configs/channel_12.yaml
    #   - components/wavinahc9000v2/configs/channel_13.yaml 
    #   - components/wavinahc9000v2/configs/channel_14.yaml
    #   - components/wavinahc9000v2/configs/channel_15.yaml 
    #   - components/wavinahc9000v2/configs/channel_16.yaml 
    refresh: 0s

uart:
  - id: uart_${device}
    rx_pin: GPIO36
    tx_pin: GPIO4
    baud_rate: 38400
    stop_bits: 1
    parity: NONE
    
modbus:
    - id: ${device}_modbus
      flow_control_pin: 5
      uart_id: uart_${device}

modbus_controller:
  id: ${device}_modbus_controller
  address: 1
  modbus_id: wavin_modbus
  update_interval: 5s`

heinekmadsen, 2023-01-12:
Take a look at the dev branch… there you can do some channel alightments

heinekmadsen, 2025-09-25:
Please try https://github.com/heinekmadsen/esphome_wavinahc9000v3 and see its still an issue