heinekmadsen/esphome_components #20: Question: do you have any experience with two linked Wavin AHC 9000 controllers

jgrau, 2022-09-23:
Hi

First of all thanks for this impressive work. I currently have two separate wavin controllers connected via modbus. I can control both systems from one display connected to one of the controllers. If I remove the display and I remove that modbus cable and plug into my esp32 connected to one of the wavin controllers it works just fine. If I connect the two controllers together, fun things start to happen: the values for the different channels start to "jump". Eg. channel 1 would show a Currently temperature of 19 and then jump to 20 and then back to 19. I suspect that it's because the esp reads two different values for the same channel id(?).

Anyways: I was wondering if you have had a chance to work with two controllers linked by cable? Or if you believe there's a way to get this code working for two controllers. If not, then I'll just install a separate esp for the each of the controllers.

heinekmadsen, 2022-09-29:
Hi...
I only have one Wavin Ahc 9000 controller so i'm not able to test it out.
Looking at what time has been put into the current code, I would go for the extra esp :D

jgrau, 2022-09-29:
Thanks for the reply. I'll go for an extra esp.

closed 2022-09-29 by jgrau