heinekmadsen/esphome_components #23: Maybe wrong "virkningsgrad" / efficiency

sknsean, 2022-10-28:
https://github.com/heinekmadsen/esphome_components/blob/d41fb804eb972038df6eddfd0395c6f5f76800e2/components/genvexv2/optima250.yaml#L49

If I remember correcty(from the manual or genvex webpage) the real formula for efficiency is:
float x = (100 * (genvex_temp_t1_inlet - genvex_temp_t3_outdoor) / (genvex_temp_t7_extract - genvex_temp_t4_exhaust))

heinekmadsen, 2022-10-28:
If you send me the documentation i'll be happy to Update it

sknsean, 2022-10-29:
This agree's with me:
https://www.lav-det-selv.dk/forum/t/214950/s/genvex-energy-2-ydelse

But this agree's with the current code:
https://www.exhausto.dk/~/media/Global/PDF/Institute/Building_ventilation/performancetest_ventilation.pdf page 7

Lets keep the current code :D

closed 2022-10-29 by sknsean