heinekmadsen/esphome_components #24: pin connections? :)

McAndersDK, 2022-12-11:
Am i missing something or dont this repo have info about how to connect the board to the genvex? :)
I have been looking into the following:
Optima 250 and Genvex GE Energy 2
The genvex GE Energy 2 have the following connectors on the mainboard to the Optima 250:
L2
[image]
and from the genvex connect documentation I can see the pin outs for the Optima 250:
[image]
but im still not sure what to connect to what? :)

heinekmadsen, 2023-03-17:
Did you get this working?

McAndersDK, 2023-03-17:
no not yet.

heinekmadsen, 2023-03-17:
Well... you need a modbus module.
ESP32 -> Modbus module -> Optima
Easiest is to buy a complete module.
I believe this will work, cutting one end of the RJ45 cable and connect the wires to the optima display.
Not sure if it can power the device from the optima display, or if power using the USB port is needed
https://ustepper.com/shop/home/18-esphome-modbus-module.html

McAndersDK, 2023-11-02:
Hey again,
so is this not good enought? https://arduinotech.dk/shop/ttl-til-rs-485-modul/

heinekmadsen, 2023-11-02:
Not really sure. I’m, not that into the hw part.

Try asking in the Danish Home Assistant or Nilan/Home Assistant facebook groups