heinekmadsen/esphome_components #30: Duplicate modbus command found

pvprodk, 2023-10-23:
I'm using the esp32-c3-devkitm-1 hardware from @nic6911 and upgrading from wavinahc9000 to wavinahc9000v2, but with v2 I get the following errors in the log.

My UART configuration is as specified by nic6911:
```
uart:
  - id: uart_${device}
    rx_pin: 20
    tx_pin: 21
    baud_rate: 38400
    stop_bits: 1
    parity: NONE
```

Modbus communication was working fine with the old wavinahc9000 code, any advice?

[image]

heinekmadsen, 2023-10-23:
Please share your yaml

[image]<https://user-images.githubusercontent.com/79306514/277470759-705b8bbb-576e-422c-a5d4-d4f9af357882.png>

pvprodk, 2023-10-23:
```
substitutions:
  device: wavin
  name: Wavin
  
  # CHANNEL friendly names (If using spaces remember to add ")
  channel_01_friendly_name: Leonora
  channel_02_friendly_name: Vincent
  channel_03_friendly_name: "Bad St."
  channel_04_friendly_name: "Bad 1st"
  channel_05_friendly_name: Fordelergang
  channel_06_friendly_name: Forgang
  channel_07_friendly_name: Stue-ost
  channel_08_friendly_name: Stue-vest
  channel_09_friendly_name: Kokken
  channel_10_friendly_name: Garage
  channel_11_friendly_name: "Channel 11"
  channel_12_friendly_name: "Channel 12"
  channel_13_friendly_name: "Channel 13"
  channel_14_friendly_name: "Channel 14"
  channel_15_friendly_name: "Channel 15"
  channel_16_friendly_name: "Channel 16"

  # CHANNEL ID´S (ONLY LOWER CASE LETTERS, NO SPACES) used for entity ids
  channel_01_id: leonora
  channel_02_id: vincent
  channel_03_id: bad_st
  channel_04_id: bad_1st
  channel_05_id: fordelergang
  channel_06_id: forgang
  channel_07_id: stue_ost
  channel_08_id: stue_vest
  channel_09_id: kokken
  channel_10_id: garage
  channel_11_id: channel_11
  channel_12_id: channel_12
  channel_13_id: channel_13
  channel_14_id: channel_14
  channel_15_id: channel_15
  channel_16_id: channel_16

    # CHANNEL Alignments (ONLY change if you have offsets in channel vs temperature. Some have reported this if using multiple strings pr. room thermostat)
  channel_01_sensor: "0x00"
  channel_02_sensor: "0x01"
  channel_03_sensor: "0x02"
  channel_04_sensor: "0x03"
  channel_05_sensor: "0x04"
  channel_06_sensor: "0x05"
  channel_07_sensor: "0x06"
  channel_08_sensor: "0x07"
  channel_09_sensor: "0x08"
  channel_10_sensor: "0x09"
  channel_11_sensor: "0x0A"
  channel_12_sensor: "0x0B"
  channel_13_sensor: "0x0C"
  channel_14_sensor: "0x0D"
  channel_15_sensor: "0x0E"
  channel_16_sensor: "0x0F"

  channel_01: "0x00"
  channel_02: "0x01"
  channel_03: "0x02"
  channel_04: "0x03"
  channel_05: "0x04"
  channel_06: "0x05"
  channel_07: "0x06"
  channel_08: "0x07"
  channel_09: "0x08"
  channel_10: "0x09"
  channel_11: "0x0A"
[... 77 lines cut ...]
      #- components/wavinahc9000v2/configs/channel_14_comfort.yaml # Only for channels with thermostat with IR sensor
      #- components/wavinahc9000v2/configs/channel_15.yaml 
      #- components/wavinahc9000v2/configs/channel_15_comfort.yaml # Only for channels with thermostat with IR sensor
      #- components/wavinahc9000v2/configs/channel_16.yaml 
      #- components/wavinahc9000v2/configs/channel_16_comfort.yaml # Only for channels with thermostat with IR sensor
    refresh: 0s

uart:
  - id: uart_${device}
    rx_pin: 20
    tx_pin: 21
    baud_rate: 38400
    stop_bits: 1
    parity: NONE
    
modbus:
    - id: ${device}_modbus
      uart_id: uart_${device}

modbus_controller:
  id: ${device}_modbus_controller
  address: 1
  modbus_id: wavin_modbus
  update_interval: 5s
```

heinekmadsen, 2023-10-23:
@nic6911 any idea why @pvprodk gets this on your hw?

pvprodk, 2023-10-23:
By the way, this line of code is not in your V2 [guide](https://github.com/heinekmadsen/esphome_components/blob/main/README.md#v2---wavin), but I guess it should be added to enable proper function of channel 1?

[image]

heinekmadsen, 2023-10-24:
> By the way, this line of code is not in your V2 [guide](https://github.com/heinekmadsen/esphome_components/blob/main/README.md#v2---wavin), but I guess it should be added to enable proper function of channel 1?
> 
> [image]

Thanks for the heads-up :)

nic6911, 2023-10-24:
> @nic6911 any idea why @pvprodk gets this on your hw?

At least under modbus you are missing: flow_control_pin: 10

pvprodk, 2023-10-24:
> > @nic6911 any idea why @pvprodk gets this on your hw?
> 
> At least under modbus you are missing: flow_control_pin: 10

That did the trick!
It is now working perfectly!

@heinekmadsen / @nic6911  perhaps you should include this in the V2 guide :)

heinekmadsen, 2023-10-24:
> > > @nic6911 any idea why @pvprodk gets this on your hw?
> > 
> > 
> > At least under modbus you are missing: flow_control_pin: 10
> 
> That did the trick! It is now working perfectly!
> 
> @heinekmadsen / @nic6911 perhaps you should include this in the V2 guide :)

Well.. it's hardware specific, so... i don't need the flow control on the hw i use.

closed 2023-10-24 by heinekmadsen