heinekmadsen/esphome_components #31: 2023.10.3 incompatibility

MTrab, 2023-10-24:
Cannot compile on ESPHome 2023.10.3, getting ALOT of errors.

[logs_wavin_run.txt](https://github.com/heinekmadsen/esphome_components/files/13111805/logs_wavin_run.txt)

heinekmadsen, 2023-10-24:
@MTrab Did you check the newly added info about breaking change?
Committed some changes last night that is a breaking change, but makes it possible to align channel numbers if your thermostat are not showing the expected temperatures.

https://github.com/heinekmadsen/esphome_components#recent-updates

MTrab, 2023-10-24:
Nope, didn't see those - will test later today

MTrab, 2023-10-24:
That did the trick - strange it haven't been an issue so far.
Sorry for the disturbance :D

closed 2023-10-24 by MTrab