heinekmadsen/esphome_components #35: Hardware requirements

MathiasReker, 2024-03-06:
Hello.

I am new in the field of esp32. Can any of you share what hardware you use  to run this project.

I already have an Optima 250 installed.

Jogge, 2024-03-06:
Hello,

To run this project, you'll need a few pieces of hardware.

First, you need to verify if your Optima 250 display supports Modbus. You can check this on the back of the display. Here's an image for reference:

[image]

Also, ensure that your Optima 250 Display has the Modbus settings configuration in the software version you are using. You can find these settings on the last page of this document: [Optima 250 Modbus Settings](https://github.com/lskov/esphome_components/blob/main/components/genvexv2/doc/Betjeningsvejledning_Modbus_Communication_OPT250_V3.1_20.11.14_EN.pdf)

In addition to the Optima 250 display, you'll need an ESP32. You will also require a RS485 converter. I personally use this: [TTL to RS485 Converter](https://www.aliexpress.com/item/32846149743.html)

To complete this project, I have bought some cables and other tools. Here are the links to these (you probably won't need everything, but it's a good start if you don't have anything already):

- [Double Head Alligator Clip Jumper Cable(20cm male and 20cm female)](https://www.aliexpress.com/item/1005005963771367.html)
- [Dupont Line Male to Male + Female and Female to Female Jumper Dupont Wire Cable (KIT 120PCS)](https://www.aliexpress.com/item/1005005945668553.html)
- [Professional Electrician Wire Tool Cable Wire Stripper (PRO)](https://www.aliexpress.com/item/1005006159788378.html)
- [Insulated Heat Shrink Tubing Colorful Heat Shrinkable Tubing (RSGT-Black-280PCS)](https://www.aliexpress.com/item/1005005945120356.html)
- [Adjustable Temperature Electric Soldering Iron (EU plug 220V)](https://www.aliexpress.com/item/1005005514613719.html)

I'm using a fork of this repo by @lskov:
https://github.com/lskov/esphome_components/tree/main/components/genvexv2

Lastly, @lskov has a comprehensive post on the "Dansk Home Assistant" Facebook group, showcasing all the options you have. You can check it out [here](https://www.facebook.com/photo/?fbid=10159873672073051).

I hope this helps you get started with your project.

MathiasReker, 2024-03-06:
> Hello,
>
> To run this project, you'll need a few pieces of hardware.
>
> First, you need to verify if your Optima 250 display supports Modbus. You can check this on the back of the display. Here's an [...]

Thank you a lot @Jogge !

Jogge, 2024-03-13:
@MathiasReker 
[image]

You need to check if there is a black component in here, like the one in my previous post.

MathiasReker, 2024-03-13:
@Jogge the black box is missing. I'm sorry it's not clear in the photo - I should have taken the photo from the same angle as you.

I have a micro USB on my Optima 250. Would I be able to convert micro USB to USB A and then into this device:
https://www.fruugo.dk/rs485-til-usb-adapter-understotter-115200-fejlfinding-af-serielt-portsaet-med-hoj-baud-hastighed/p-114506155-241093684

[image]

MathiasReker, 2024-03-13:
Or maybe I could connect this unit to the micro USB? https://arduinotech.dk/shop/usb-to-rs485-ttl-serial-converter-adapter-ft232
[image]

I could then power the ESP32 from an external power source.

heinekmadsen, 2026-02-27:
Closing as issue is stale

closed 2026-02-27 by heinekmadsen