heinekmadsen/esphome_components #44: Compile error

thomasfals, 2025-08-22:
There seems to be an issue with compile. 

```
Updating /config/esphome/genvex.yaml
------------------------------------------------------------

INFO ESPHome 2025.8.0
INFO Reading configuration /config/esphome/genvex.yaml...
INFO Updating https://github.com/heinekmadsen/esphome_components@main
INFO Updating https://github.com/heinekmadsen/esphome_components@main
INFO Generating C++ source...
INFO Compiling app...
Processing genvex (board: esp32dev; framework: arduino; platform: https://github.com/pioarduino/platform-espressif32/releases/download/54.03.21-2/platform-espressif32.zip)
--------------------------------------------------------------------------------
INFO Package configuration completed successfully
INFO Package configuration completed successfully
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-arduinoespressif32 @ 3.2.1 
 - framework-arduinoespressif32-libs @ 5.4.0+sha.858a988d6e 
 - tool-esptoolpy @ 5.0.2 
 - tool-mklittlefs @ 3.2.0 
 - toolchain-xtensa-esp-elf @ 14.2.0+20241119
Dependency Graph
|-- Networking @ 3.2.1
|-- AsyncTCP @ 3.4.5
|-- WiFi @ 3.2.1
|-- FS @ 3.2.1
|-- Update @ 3.2.1
|-- ESPAsyncWebServer @ 3.7.10
|-- ESP32 Async UDP @ 3.2.1
|-- DNSServer @ 3.2.1
|-- ESPmDNS @ 3.2.1
|-- noise-c @ 0.1.10
Compiling .pioenvs/genvex/src/esphome/components/genvexv2/climate/genvexv2_climate.cpp.o
Compiling .pioenvs/genvex/src/esphome/components/ota/ota_backend_arduino_esp32.cpp.o
Compiling .pioenvs/genvex/src/esphome/components/safe_mode/safe_mode.cpp.o
Compiling .pioenvs/genvex/src/esphome/components/select/select.cpp.o
Compiling .pioenvs/genvex/src/esphome/components/select/select_call.cpp.o
Compiling .pioenvs/genvex/src/esphome/components/select/select_traits.cpp.o
Compiling .pioenvs/genvex/src/esphome/components/sensor/automation.cpp.o
Compiling .pioenvs/genvex/src/esphome/components/sensor/filter.cpp.o
Compiling .pioenvs/genvex/src/esphome/components/sensor/sensor.cpp.o
Compiling .pioenvs/genvex/src/esphome/components/socket/bsd_sockets_impl.cpp.o
Compiling .pioenvs/genvex/src/esphome/components/socket/socket.cpp.o
Compiling .pioenvs/genvex/src/esphome/components/status/status_binary_sensor.cpp.o
Compiling .pioenvs/genvex/src/esphome/components/template/sensor/template_sensor.cpp.o
src/esphome/components/genvexv2/climate/genvexv2_climate.cpp: In member function 'virtual void esphome::genvexv2::Genvexv2Climate::control(const esphome::climate::ClimateCall&)':
src/esphome/components/genvexv2/climate/genvexv2_climate.cpp:61:51: error: no matching function for call to 'to_string(const char [2])'
   61 |         this->custom_fan_mode = esphome::to_string("2");
      |                                 ~~~~~~~~~~~~~~~~~~^~~~~
In file included from /data/cache/platformio/packages/toolchain-xtensa-esp-elf/xtensa-esp-elf/include/c++/14.2.0/string:54,
                 from src/esphome/core/component.h:6,
                 from src/esphome/components/genvexv2/climate/genvexv2_climate.h:3,
                 from src/esphome/components/genvexv2/climate/genvexv2_climate.cpp:1:
/data/cache/platformio/packages/toolchain-xtensa-esp-elf/xtensa-esp-elf/include/c++/14.2.0/bits/basic_string.h:4327:3: note: candidate: 'std::string std::__cxx11::to_string(long long unsigned int)' (near match)
 4327 |   to_string(unsigned long long __val)
      |   ^~~~~~~~~
/data/cache/platformio/packages/toolchain-xtensa-esp-elf/xtensa-esp-elf/include/c++/14.2.0/bits/basic_string.h:4327:3: note:   conversion of argument 1 would be ill-formed:
src/esphome/components/genvexv2/climate/genvexv2_climate.cpp:61:52: error: invalid conversion from 'const char*' to 'long long unsigned int' [-fpermissive]
   61 |         this->custom_fan_mode = esphome::to_string("2");
      |                                                    ^~~
      |                                                    |
      |                                                    const char*
/data/cache/platformio/packages/toolchain-xtensa-esp-elf/xtensa-esp-elf/include/c++/14.2.0/bits/basic_string.h:4310:3: note: candidate: 'std::string std::__cxx11::to_string(long long int)' (near match)
 4310 |   to_string(long long __val)
      |   ^~~~~~~~~
/data/cache/platformio/packages/toolchain-xtensa-esp-elf/xtensa-esp-elf/include/c++/14.2.0/bits/basic_string.h:4310:3: note:   conversion of argument 1 would be ill-formed:
src/esphome/components/genvexv2/climate/genvexv2_climate.cpp:61:52: error: invalid conversion from 'const char*' to 'long long int' [-fpermissive]
   61 |         this->custom_fan_mode = esphome::to_string("2");
      |                                                    ^~~
[... 39 lines cut ...]
 4421 |   to_string(long double __val)
      |   ^~~~~~~~~
/data/cache/platformio/packages/toolchain-xtensa-esp-elf/xtensa-esp-elf/include/c++/14.2.0/bits/basic_string.h:4421:25: note:   no known conversion for argument 1 from 'const char [2]' to 'long double'
 4421 |   to_string(long double __val)
      |             ~~~~~~~~~~~~^~~~~
/data/cache/platformio/packages/toolchain-xtensa-esp-elf/xtensa-esp-elf/include/c++/14.2.0/bits/basic_string.h:4411:3: note: candidate: 'std::string std::__cxx11::to_string(double)'
 4411 |   to_string(double __val)
      |   ^~~~~~~~~
/data/cache/platformio/packages/toolchain-xtensa-esp-elf/xtensa-esp-elf/include/c++/14.2.0/bits/basic_string.h:4411:20: note:   no known conversion for argument 1 from 'const char [2]' to 'double'
 4411 |   to_string(double __val)
      |             ~~~~~~~^~~~~
/data/cache/platformio/packages/toolchain-xtensa-esp-elf/xtensa-esp-elf/include/c++/14.2.0/bits/basic_string.h:4401:3: note: candidate: 'std::string std::__cxx11::to_string(float)'
 4401 |   to_string(float __val)
      |   ^~~~~~~~~
/data/cache/platformio/packages/toolchain-xtensa-esp-elf/xtensa-esp-elf/include/c++/14.2.0/bits/basic_string.h:4401:19: note:   no known conversion for argument 1 from 'const char [2]' to 'float'
 4401 |   to_string(float __val)
      |             ~~~~~~^~~~~
*** [.pioenvs/genvex/src/esphome/components/genvexv2/climate/genvexv2_climate.cpp.o] Error 1
========================= [FAILED] Took 10.35 seconds =========================
=========== [ERROR] /config/esphome/genvex.yaml ===========

======================== [SUMMARY] ========================
  - /config/esphome/esphome-web-9fa28c.yaml: FAILED
  - /config/esphome/genvex.yaml: FAILED
```

heinekmadsen, 2025-08-22:
Please share yaml

thomasfals, 2025-08-22:
[yaml.yml](https://github.com/user-attachments/files/21935747/yaml.yml)

heinekmadsen, 2025-09-25:
Please try https://github.com/heinekmadsen/esphome_wavinahc9000v3 and see its still an issue